esphome/esphome-flasher #1: Installing esphomeflasher-1.0.1-windows does not work

pvdw0310, 2019-02-05:
I installed esphomeflasher for windows on a windows 7 laptop and on a windows 10 desktop and that works fine. I like to install it on another windows 10 laptop and I get a error : Fatal error detected  Failes to execute script_main_ . Any idee what went wrong? Thanks.

OttoWinter, 2019-02-05:
Please check it is not a 32-bit machine. Those are not supported with the binary you downloaded

celeroll, 2019-04-30:
I have the same issue on one windows 10 64 Bit computer. Don't know what is missing. Maybe python?

OttoWinter, 2019-05-01:
@celeroll Python pre-packaged in the binary (via pyinstaller). Don't know how to debug this, maybe someone with more knowledge of pyinstaller can help with getting better logs.