esphome/esphome-flasher #2: Flashing file returns error: "has no attribute 'erase_all'"

NicoBijl, 2019-02-13:
When running esphomeflasher on my macbook, i get the following error: 

Console output in the GUI: 
 ```
Using '/dev/cu.usbserial-A4012SPV' as serial port.
Connecting....
Detecting chip type... ESP8266
Connecting...

Chip Info:
 - Chip Family: ESP8266
 - Chip Model: ESP8285
 - Chip ID: 0082C715
 - MAC Address: DC:4F:22:82:C7:15
Uploading stub...
Running stub...
Stub running...
 - Flash Size: 1MB
 - Flash Mode: qio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.5s
Unexpected error: 'MockEsptoolArgs' object has no attribute 'erase_all'
```

Terminal stacktrace: 
```
Exception in thread Thread-7:
Traceback (most recent call last):
  File "/usr/local/Cellar/python/3.7.2_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.7/site-packages/esphomeflasher/gui.py", line 48, in run
    run_esphomeflasher(argv)
  File "/usr/local/lib/python3.7/site-packages/esphomeflasher/__main__.py", line 113, in run_esphomeflasher
    esptool.write_flash(stub_chip, mock_args)
  File "/usr/local/bin/esptool.py", line 2128, in write_flash
    if args.erase_all:
AttributeError: 'MockEsptoolArgs' object has no attribute 'erase_all'```

Any ideas how to fix this?

NicoBijl, 2019-02-13:
Flashing the same .bin file, with esphomeflasher on windows does work..

bfiset, 2019-02-14:
Does not work for me on Windows 10 Pro.

I get this error.

bennydiamond, 2019-02-14:
Does not work for me on Windows 10 Pro.

I get this error trying to flash my generated bin.

novikovav69, 2019-02-16:
I have a similar problem with esp32
[image]
Linuxmint

OttoWinter, 2019-05-01:
This should be fixed in the 1.0.2 release.

closed 2019-05-01 by OttoWinter

legomind, 2019-05-02:
I just tried the 1.0.2 release, and I am still getting the same error.

bennydiamond, 2019-05-15:
Just tried latest commit in master branch (1.0.2) and I have the following error:

> Unexpected error: 'MockEsptoolArgs' object has no attribute 'erase_all'

Using Windows 10 Pro and a Wemos D1 Mini

here is the full log transcript:

> Using 'COM3' as serial port.
> Connecting....
> Detecting chip type... ESP8266
> Connecting....
> 
> Chip Info:
>  - Chip Family: ESP8266
>  - Chip Model: ESP8266EX
>  - Chip ID: 0085F65A
>  - MAC Address: 84:0D:8E:85:F6:5A
> Uploading stub...
> Running stub...
> Stub running...
>  - Flash Size: 4MB
>  - Flash Mode: dout
>  - Flash Frequency: 40MHz
> Erasing flash (this may take a while)...
> Chip erase completed successfully in 9.7s
> Unexpected error: 'MockEsptoolArgs' object has no attribute 'erase_all'