esphome/esphome-flasher #6: Windows thinks 1.2.0 x86 has a virus - cannot download

borpin, 2019-11-02:
https://github.com/esphome/esphome-flasher/releases/download/v1.2.0/ESPHome-Flasher-1.2.0-windows-x86.exe

64 bit is OK.

OttoWinter, 2019-11-02:
I'm going to assume that's a false positive.

The binary is generated on Github Actions ([here](https://github.com/esphome/esphome-flasher/blob/master/.github/workflows/build.yml#L33)) - and I trust GitHub.

An online scan of the binary also says there's nothing found: https://www.virustotal.com/gui/url/1a0f82e9ffde9f9c1983628aaf4b93013863918768b1f40e59ef6296ad854c35/detection

I think the windows defender works with fingerprinting - maybe there has been a virus somewhere that also used PyInstaller, and the fingerprint match is too broad. Or they have some sort of detection for embedded ELF files (like the ESP32/8266 bootloader). Either way, it's a black box.

OttoWinter, 2019-11-02:
Going to close this - from the negative online scan and that we can't really do anything against a virus checker having false positive.

closed 2019-11-02 by OttoWinter

borpin, 2019-11-02:
If anyone finds this, You need to go into Windows Defender and allow the 'threat'.