esphome/esphome-flasher #10: error esphomeflasher

chusrufo, 2019-11-23:
Hi
from a mac, when trying to load the .bin of the esp32 in esphomeflasher, it gives me the following error:

Using '/dev/cu.SLAB_USBtoUART' as serial port.
Connecting........___
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 160MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 84:0D:8E:12:02:A4
Uploading stub...
Running stub...
Stub running...
 - Flash Size: 4MB
Unexpected error: Error while retrieving firmware file 'https://raw.githubusercontent.com/espressif/arduino-esp32/1.0.0/tools/sdk/bin/bootloader_dio_40m.bin': HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /espressif/arduino-esp32/1.0.0/tools/sdk/bin/bootloader_dio_40m.bin (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x1147a20b8>: Failed to establish a new connection: [Errno 61] Connection refused',))

terminal code:

Last login: Sat Nov 23 16:50:21 on ttys000
MacBook-Pro-de-Jesus:~ chusrufo$ esphomeflasher
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/urllib3/connection.py", line 157, in _new_conn
    (self._dns_host, self.port), self.timeout, **extra_kw
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/urllib3/util/connection.py", line 84, in create_connection
    raise err
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/urllib3/util/connection.py", line 74, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 61] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/urllib3/connectionpool.py", line 672, in urlopen
    chunked=chunked,
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/urllib3/connectionpool.py", line 376, in _make_request
    self._validate_conn(conn)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/urllib3/connectionpool.py", line 994, in _validate_conn
    conn.connect()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/urllib3/connection.py", line 334, in connect
    conn = self._new_conn()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/urllib3/connection.py", line 169, in _new_conn
    self, "Failed to establish a new connection: %s" % e
urllib3.exceptions.NewConnectionError: <urllib3.connection.VerifiedHTTPSConnection object at 0x1154560b8>: Failed to establish a new connection: [Errno 61] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/urllib3/connectionpool.py", line 720, in urlopen
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/urllib3/util/retry.py", line 436, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /espressif/arduino-esp32/1.0.0/tools/sdk/bin/bootloader_dio_40m.bin (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x1154560b8>: Failed to establish a new connection: [Errno 61] Connection refused',))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/esphomeflasher/common.py", line 140, in open_downloadable_binary
    response = requests.get(path)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/requests/api.py", line 75, in get
    return request('get', url, params=params, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /espressif/arduino-esp32/1.0.0/tools/sdk/bin/bootloader_dio_40m.bin (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x1154560b8>: Failed to establish a new connection: [Errno 61] Connection refused',))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/esphomeflasher/gui.py", line 163, in run
    run_esphomeflasher(argv)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/esphomeflasher/__main__.py", line 118, in run_esphomeflasher
    args.otadata)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/esphomeflasher/common.py", line 174, in configure_write_flash_args
    format_bootloader_path(bootloader_path, flash_mode, flash_freq))
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/esphomeflasher/common.py", line 147, in open_downloadable_binary
    "Error while retrieving firmware file '{}': {}".format(path, err))
esphomeflasher.common.EsphomeflasherError: Error while retrieving firmware file 'https://raw.githubusercontent.com/espressif/arduino-esp32/1.0.0/tools/sdk/bin/bootloader_dio_40m.bin': HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /espressif/arduino-esp32/1.0.0/tools/sdk/bin/bootloader_dio_40m.bin (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x1154560b8>: Failed to establish a new connection: [Errno 61] Connection refused',))

OttoWinter, 2019-11-26:
Hi, you need to have an internet connection while uploading for ESP32 (the bootloader needs to be downloaded from github, as packing it in with pyinstaller directly is not easy).

closed 2019-11-26 by OttoWinter

remzouz64, 2021-08-04:
Bonjour,
impossible de flasher mon ESP32. Est ce que quelqu'un a une solution ?
Merci d'avance