esphome/esphome-flasher #11: wont find the NodeMCU's Serial port

Amtech7, 2019-11-26:
Not sure if this is just an issue with my quite yet. I've been trying to troubleshoot, new chord, driver. but wanted to check if anyone was having the issue here.

mcpacific, 2019-11-27:
I have the same issue. I tried a different Micro USB cord and same result.

OttoWinter, 2019-11-29:
ESPHome-Flasher displays exactly what the operating system sees.

So the only reason the serial port is not detected would be because the OS does not find it.

There may be many reasons for this, including not having the drivers, wrong cables etc. But that's an issue of the OS, not esphome-flasher.

If you do find a solution, please post it here so that it could potentially be added to the docs.

closed 2019-11-29 by OttoWinter