esphome/esphome-flasher #13: Small issue in Console window

stefanroelofs, 2019-12-22:
The log window is the black area in esphome-flasher where the log messages are printed.

When I flash an esp8266, I see a log message that indicated the progress of flashing. 
Something like this: Writing at 0x00048000... (100 %)

This log message eats up all other log messages above it. 
So this log message moves up and deletes the messages above it.

OttoWinter, 2020-02-09:
Please add what platform this is on (windows etc).

I do not see this happening on my system.

stefanroelofs, 2020-02-09:
Im using ESPHome-Flasher-1.2.0-windows-x64
On a Windows 10 laptop
Flashing a Shelly1 device.

Here's a small clip of it happening:
https://youtu.be/VhshUZu4iE0

sermayoral, 2020-07-19:
I have the same problem. Windows 10 x64 with all ESPs. Its a minor problem,..

OttoWinter, 2020-07-29:
Hmm, don't know what it is; I checked with a windows 10 VM and it worked there. I'll try it on a real windows laptop when i get access to one (probably around next week)

sermayoral, 2020-07-29:
@OttoWinter is an small issue, but i have recorded a video to show you the problem: https://youtu.be/m32L21GU8tE

Thanks!

DavidDeSloovere, 2020-08-04:
I'm also seeing this on a recent Windows 10 build.
But flashing works, it's just cosmetics.
Keep up the great work Otto.

drogfild, 2020-10-16:
I have had this also always with ESPHome Flasher. Now using Flasher 1.3.0 with Win10 ver 2004. Can it be that there is char in the "Writing" line that makes it to misbehave? Small issue, but would be nice to see whole log after flashing.
All other logging and flashing with this tool works perfectly. Really nice tool!

sermayoral, 2021-03-19:
I think bumping esptool library to 3.0 will do the trick!

closed 2021-11-06 by github-actions[bot]