esphome/esphome-flasher #14: Permission Error 13

Tom-Neverwinter, 2019-12-22:
Unexpected error: could not open port 'COM4': PermissionError(13, 'Access is denied.', None, 5)

windows 10, port is selected, application was run as admin

meteorinca, 2020-01-04:
Had the same issue. Was working a few days ago. I fixed it by installing the usb-to-uart-bridge-vcp-drivers for Windows and rebooting the machine. It now works both through cmd and the GUI.

OttoWinter, 2020-02-09:
Since this appears to be a driver issue and not esphomeflasher-specific, I will close.

closed 2020-02-09 by OttoWinter

sabellwind, 2020-02-29:
Hey @OttoWinter 

I'm still experiencing this issue even after updating drivers, I wrote up a bit on the Hass forums

https://community.home-assistant.io/t/flashing-wemos-d1-mini-with-esphome-unexpected-error-could-not-open-port-com4-permissionerror-13-access-is-denied-none-5/175361?u=sabellwind

jwodev, 2020-10-10:
Hi there,
I had the same issue, but it only occured when I pressed [Flash ESP] after I had pressed [View Logs]
When I start the tool, connect the ESP via FTDI232 (in my case ESP-32 IO0 grounded), select the serial port, select my bin file and then directly push [Flash ESP] it flashed without any problem.

daes10, 2021-04-30:
I have the same issue like @jwodev and now without pressing the "view log" tab it worked for me.

Tom-Neverwinter, 2021-05-01:
Solution was to use a different cable. 

The anker powerline I was using has a resistor that prevented it from writing to the board.

SimonVleugels, 2021-08-25:
Not using an Anker cable did it for me.