esphome/esphome-flasher #15: "Timed out waiting for packet header" while flashing M5Stack Atom

timmmmmmmmm, 2020-01-24:
I tried to flash a bin file using esphome-flasher 1.2.0 to my [M5Stack Atom](https://m5stack.com/products/atom-matrix-esp32-development-kit), but was getting the "Timed out waiting for packet header" after getting the basic chip identification data in the log. Both on macOS and Windows.

Flashing succeeded using esphome-flasher 1.1.0. I expect it has to do with the faster upload speed in 1.2.0.

tathamoddie, 2020-02-08:
Experience the same issue here with 1.2.0 and M5Stack Atom Lite.

Works when reverting back to 1.1.0.

mcflexus, 2020-02-27:
Experiencing the same issue here with 1.2.0 and M5Stack Atom Lite.

Works when reverting back to 1.1.0.

i00, 2020-03-02:
Same here works with 1.1.0

Log:

Using 'COM3' as serial port.
Connecting........___
Detecting chip type... ESP8266
Connecting.....

Chip Info:
 - Chip Family: ESP8266
 - Chip Model: ESP8266EX
 - Chip ID: 00448915
 - MAC Address: BC:DD:C2:44:89:15
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Unexpected error: Reading chip details failed: Timed out waiting for packet header

dogcatfish, 2020-07-10:
Same here with Atom Matrix.  "Unexpected error: Reading chip details failed: Timed out waiting for packet header".
1.1.0 release worked fine for that too.

closed 2020-07-13 by OttoWinter

i00, 2020-07-19:
@OttoWinter  ... why was this closed ... also older releases work as stated

MaartenKleijwegt, 2020-10-19:
I'm still experiencing this issue with a m5stickc. Release 1.1 works while 1.3 doesn't

Klarkin2, 2020-12-07:
I'm experiencing this issue with 1.3 as well