esphome/esphome-flasher #17: IMAC not detecting board using ESPHOME-FLASHER

bakaufman, 2020-02-09:
Man I hope this is a stupid mistake on my end. I DL the latest version of flasher on my iMAC running Catalina. It doesn't detect the device on the serial port (ESP32 dev board). Otherwise I have my compile bin ready to go. Is there a dependency that needs to be mentioned in the instructions?

Jaykurb, 2020-02-11:
> Man I hope this is a stupid mistake on my end. I DL the latest version of flasher on my iMAC running Catalina. It doesn't detect the device on the serial port (ESP32 dev board). Otherwise I have my compile bin ready to go. Is there a dependency that needs to be mentioned in the instructions?

I have the same issue but with 16" Macbook pro. Oddly this issue does not occur on my older 15" Macbook pro. The only work around for me atm is to use a virtual machine and use windows.

OttoWinter, 2020-07-29:
This is a driver issue - not ESPHome-Flasher related. In general if a port does not show up that means there's an issue with the operating system and missing or broken drivers.

I think there was an issue some time ago with an update to the siliconlabs VCP CP210X macOS driver together with a specific macOS version. Please try to update that to the latest version.

closed 2020-07-29 by OttoWinter