esphome/esphome-flasher #18: Unexpected error: Reading chip details failed: Timed out waiting for packet header

zejulio, 2020-03-09:
Hello, I'm using M5Stick C and when I try to flash with esphome-flasher in windows, I've got this:

Connecting.....
Detecting chip type... ESP32
Connecting.....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-PICO-D4 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: YES
 - Has Factory-Calibrated ADC: YES
 - MAC Address: D8:A0:1D:68:CF:18
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Unexpected error: Reading chip details failed: Timed out waiting for packet header

Anyone can help?
Thanks!

sander816, 2020-04-23:
yes, same here with esp01:

Using 'COM6' as serial port.
Connecting....
Detecting chip type... ESP8266
Connecting...

Chip Info:
 - Chip Family: ESP8266
 - Chip Model: ESP8266EX
 - Chip ID: 00157A04
 - MAC Address: CC:50:E3:15:7A:04
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Unexpected error: Error changing ESP upload baud rate: Timed out waiting for packet header

sander816, 2020-04-23:
this esp01 used to flash correct with esphome

misosec, 2020-07-30:
I am using M5 Stack Atom Lite and I have the same error message.
Did anybody resolve this issue?

JacobD10, 2020-08-05:
I was having this same issue on v1.3.0, I tried v1.2.0 and still same issue.
After trying v1.1.0 it worked (although there was some errors in displaying the responses on screen).

ageev, 2020-10-08:
yep, M5 Atom, 1.1.0 worked perfectly with no errors.

Tuckie, 2020-12-01:
Just wanted to report the same issue on two different M5 Stack Atom Lites as well. Switching to 1.1.0 worked!

wenzhoou, 2020-12-10:
Same issue on v1.3.0. I am use espino. 
And got correct o After back to v1.1.0.  thanks JacobD10.

gordon-zhao, 2020-12-26:
I believe the problem is caused by forcing the board to use the unsupported baud rate. The error occurs on my M5Stack Atom Lite right after the line "Changing baud rate to 460800" showing up in the log, which make sense since that board only supports up to 115200 baud rate. FYI, this *feature* is introduced in v1.2.0, that explains why v1.1.0 works.

birchje, 2021-01-08:
Had the same problem in Arduino IDE whe trying to flash the ESP32 CAM, solved by setting the baud rate to 115200 in \ArduinoData\packages\esp32\hardware\esp32\1.0.3\boards.txt
So I guess the same would work here.

#esp32cam.upload.speed=460800
esp32cam.upload.speed=115200

MisterTFM, 2021-02-28:
Is there anyway to manually slow the baud rate down when flashing a M5Stickc with ESPHome-flasher? Previous versions won't run on x86.

I keep getting the same error:

Changing baud rate to 460800
Changed.
Unexpected error: Reading chip details failed: Timed out waiting for packet header

Darcade, 2021-10-30:
Had exactly the same issue with 2 of my ESP-01. Spent already over 2 hours flashing whithout success.
Just used version 1.1 and it worked fine.

gordon-zhao, 2021-10-30:
> Had exactly the same issue with 2 of my ESP-01. Spent already over 2 hours flashing whithout success. Just used version 1.1 and it worked fine.

Which version did you use? Could you try the newest version 1.4?

redparchel, 2022-01-07:
> Which version did you use? Could you try the newest version 1.4?

version 1.4 resolved this issue for me on the atom lite. thanks!

closed 2022-04-15 by github-actions[bot]