esphome/esphome-flasher #19: Trojan in ESPHome-Flasher-1.2.0-windows-x86.exe

ppenders, 2020-03-24:
ESPHome-Flasher-1.2.0-windows-x86.exe contains  Win32/Zpevdo.A trojan

DavidDeSloovere, 2020-04-13:
@ppenders Got any info to back this up? Pretty sure Otto would not do that.

ppenders, 2020-04-17:
[image]

OttoWinter, 2020-07-29:
Duplicate of https://github.com/esphome/esphome-flasher/issues/3

Yeah I don't know where this is coming from. The build system is visible here: https://github.com/esphome/esphome-flasher/tree/master/.github/workflows - it's compiled directly on GitHub Actions, and I doubt their systems would inject trojans.

I've heard a bit in the past about some malware increasingly using python - and often python software for windows is packaged with PyInstaller, which ESPHome-Flasher also uses.
Maybe microsoft's malware filters are too broad and trigger on any PyInstaller-packaged software? I don't know, could be.

Also, as seen in #3 I uploaded the .exe to an online malware scanning service and that turned up no malware warnings, if that makes you feel better.

OttoWinter, 2020-07-29:
Closing because there's nothing I can do here, probably just some too broad malware filter :(

closed 2020-07-29 by OttoWinter