esphome/esphome-flasher #20: problem running esphome flasher in lubuntu 20.04

netweaver1970, 2020-04-25:
Hi,
I tried to get it to run, installed all dependencies and now I got this traceback when starting :

```
Traceback (most recent call last):
  File "/home/geertsteyaert/esphome-flasher-master/esphome-flasher-master/esphomeflasher/gui.py", line 286, in OnInit
    frame = MainFrame(None, "esphome-flasher (Based on NodeMCU PyFlasher)")
  File "/home/geertsteyaert/esphome-flasher-master/esphome-flasher-master/esphomeflasher/gui.py", line 177, in __init__
    self._init_ui()
  File "/home/geertsteyaert/esphome-flasher-master/esphome-flasher-master/esphomeflasher/gui.py", line 226, in _init_ui
    serial_boxsizer.Add(reload_button, 0, wx.ALIGN_RIGHT, 20)
wx._core.wxAssertionError: C++ assertion "!(flags & wxALIGN_RIGHT)" failed at /home/wxpy/wxPython-4.1.0/ext/wxWidgets/src/common/sizer.cpp(2094) in DoInsert(): Horizontal alignment flags are ignored in horizontal sizers
OnInit returned false, exiting...
```

Any idea?

Thanks.

didaccs, 2020-04-27:
I've got the same problem on ubuntu 18.04 and python 3.6

didaccs, 2020-04-28:
As a workaround I uninstalled wxPython and esphomeflasher and install previous version of wxPython (4.0.7) and install again esphomeflasher, and now works fine!

ymoona, 2020-05-02:
I can confirm that is a bug on Windows 10 as well. 
Also the work around to deploy wxPython 4.0.7 works on W10 (python 3.8.2)

damitjimii, 2020-05-13:
Just fyi: 
pip3 uninstall package wxPython
pip3 uninstall esphomeflasher
pip3 cache purge
pip3 install wxPython==4.0.7.post2
pip3 install esphomeflasher

MachinesJesus, 2020-05-28:
Thank you @damitjimii

PR #21 (2020-07-13): Fix alignment error message
Fix #20 by changing alignment value
closed 2020-07-13 by OttoWinter