esphome/esphome-flasher #22: Weird access to a non existing path...

Thanh-Viet, 2020-07-21:
I started esphomeflasher from command line on my Mac (Catalina 10.15.6 (19G73))

serial_boxsizer.Add(reload_button, 0, wx.ALIGN_RIGHT, 20)
wx._core.wxAssertionError: C++ assertion "!(flags & wxALIGN_RIGHT)" failed at /Users/robind/projects/bb2/dist-osx-py38/build/ext/wxWidgets/src/common/sizer.cpp(2098) in DoInsert(): Horizontal alignment flags are ignored in horizontal sizers
OnInit returned false, exiting...

/Users/robind/ is not on my mac...

facastagnini, 2020-07-29:
same issue on macos High Sierra 10.13.6 (17G14019)

OttoWinter, 2020-07-29:
How did you install the flasher? with `pip3`?

That will pull in the bdist_wheel (precompiled version) of the python package wxpython, and apparently the developer that maintains wxpython has the username `robind` as seen here: https://discuss.wxpython.org/t/is-wxwidgets-compiled-in-debug-mode-by-buildbot-on-purpose/34799

facastagnini, 2020-07-29:
**First the current env**

> Python 3.7.7
> pip 20.0.2

**Prereq install**
> $ pip3 install --user wxpython
> [...]
> $ pip3 list | grep -i wxpython
> wxPython           4.1.0

**Installing the pip package**
> $ pip3 install --user esphomeflasher
> Processing /Users/myusername/Library/Caches/pip/wheels/2b/2b/d5/68f35b4aeccb9df78ca714dc6b0526daea244a0fe9ffb583c9/esphomeflasher-1.2.0-py3-none-any.whl
> Requirement already satisfied: esptool==2.8 in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from esphomeflasher) (2.8)
> Requirement already satisfied: requests<3,>=2.0 in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from esphomeflasher) (2.24.0)
> Requirement already satisfied: wxpython<5.0,>=4.0 in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from esphomeflasher) (4.1.0)
> Requirement already satisfied: ecdsa in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from esptool==2.8->esphomeflasher) (0.15)
> Requirement already satisfied: pyaes in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from esptool==2.8->esphomeflasher) (1.6.1)
> Requirement already satisfied: pyserial>=3.0 in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from esptool==2.8->esphomeflasher) (3.4)
> Requirement already satisfied: chardet<4,>=3.0.2 in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from requests<3,>=2.0->esphomeflasher) (3.0.4)
> Requirement already satisfied: certifi>=2017.4.17 in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from requests<3,>=2.0->esphomeflasher) (2020.6.20)
> Requirement already satisfied: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from requests<3,>=2.0->esphomeflasher) (1.25.10)
> Requirement already satisfied: idna<3,>=2.5 in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from requests<3,>=2.0->esphomeflasher) (2.10)
> Requirement already satisfied: six in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from wxpython<5.0,>=4.0->esphomeflasher) (1.15.0)
> Requirement already satisfied: numpy; python_version >= "3.0" in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from wxpython<5.0,>=4.0->esphomeflasher) (1.17.4)
> Requirement already satisfied: pillow in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from wxpython<5.0,>=4.0->esphomeflasher) (7.2.0)
> Installing collected packages: esphomeflasher
> Successfully installed esphomeflasher-1.2.0

> esphomeflasher --help
> [...]
> $ echo $?
> 0
> $ esphomeflasher
> Traceback (most recent call last):
>   File "/Users/myusername/Library/Python/3.7/lib/python/site-packages/esphomeflasher/gui.py", line 286, in OnInit
>     frame = MainFrame(None, "esphome-flasher (Based on NodeMCU PyFlasher)")
>   File "/Users/myusername/Library/Python/3.7/lib/python/site-packages/esphomeflasher/gui.py", line 177, in __init__
>     self._init_ui()
>   File "/Users/myusername/Library/Python/3.7/lib/python/site-packages/esphomeflasher/gui.py", line 226, in _init_ui
>     serial_boxsizer.Add(reload_button, 0, wx.ALIGN_RIGHT, 20)
> wx._core.wxAssertionError: C++ assertion "!(flags & wxALIGN_RIGHT)" failed at /Users/robind/projects/bb2/dist-osx-py37/build/ext/wxWidgets/src/common/sizer.cpp(2098) in DoInsert(): Horizontal alignment flags are ignored in horizontal sizers
> OnInit returned false, exiting...

**Installing from source**
> $ pip3 uninstall esphomeflasher
> Found existing installation: esphomeflasher 1.2.0
> Uninstalling esphomeflasher-1.2.0:
>   Would remove:
>     /Users/myusername/Library/Python/3.7/bin/esphomeflasher
>     /Users/myusername/Library/Python/3.7/lib/python/site-packages/esphomeflasher-1.2.0.dist-info/*
>     /Users/myusername/Library/Python/3.7/lib/python/site-packages/esphomeflasher/*
> Proceed (y/n)? y
>   Successfully uninstalled esphomeflasher-1.2.0

> $ cd /Downloads
> $ git clone git@github.com:esphome/esphome-flasher.git
> [...]
> pip3 install --user -e .
> Obtaining file:///Users/myusername/Downloads/esphome-flasher
> Requirement already satisfied: wxpython<5.0,>=4.0 in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from esphomeflasher==1.2.0) (4.1.0)
> Requirement already satisfied: esptool==2.8 in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from esphomeflasher==1.2.0) (2.8)
> Requirement already satisfied: requests<3,>=2.0 in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from esphomeflasher==1.2.0) (2.24.0)
> Requirement already satisfied: numpy; python_version >= "3.0" in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from wxpython<5.0,>=4.0->esphomeflasher==1.2.0) (1.17.4)
> Requirement already satisfied: pillow in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from wxpython<5.0,>=4.0->esphomeflasher==1.2.0) (7.2.0)
> Requirement already satisfied: six in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from wxpython<5.0,>=4.0->esphomeflasher==1.2.0) (1.15.0)
> Requirement already satisfied: pyaes in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from esptool==2.8->esphomeflasher==1.2.0) (1.6.1)
> Requirement already satisfied: pyserial>=3.0 in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from esptool==2.8->esphomeflasher==1.2.0) (3.4)
[... cut ...]
> Requirement already satisfied: chardet<4,>=3.0.2 in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from requests<3,>=2.0->esphomeflasher==1.2.0) (3.0.4)
> Requirement already satisfied: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in /Users/myusername/Library/Python/3.7/lib/python/site-packages (from requests<3,>=2.0->esphomeflasher==1.2.0) (1.25.10)
> Installing collected packages: esphomeflasher
>   Running setup.py develop for esphomeflasher
> Successfully installed esphomeflasher

> $ esphomeflasher

that opens the UI and works as expected

closed 2022-02-05 by github-actions[bot]