esphome/esphome-flasher #25: Permission Error 13 part 2?

gijoegotyler, 2020-07-31:
I know that there was an issue with the same problem however the solution given blamed it on drivers but I have updated my drivers and programmed my esp8266 boards however when I go to change the software it seems to get the same error.

RcTomcat1, 2021-01-16:
I can confirm this.
I wanted to reflash one of my esp8266 devices and i always get an
Unexpected error: could not open port 'COM4': PermissionError(13, 'Access is denied.', None, 5) error.

Got a brand new device and was able to flash it.
Trying to reflash it gives the same error
Unexpected error: could not open port 'COM4': PermissionError(13, 'Access is denied.', None, 5)

missmah, 2022-01-19:
Hitting view logs works the first time but not after that. To flash, I need to remove the device from USB, then flash immediately, otherwise I get the same errors as the above.

closed 2022-04-27 by github-actions[bot]