esphome/esphome-flasher #26: Lates version 1.3 can not be opened

zuidwijk, 2020-08-01:
[image]

I'm working on a MacBook Pro 2019 with latest Catalina on it.

JJK801, 2020-08-02:
Same issue here

AalianKhan, 2020-08-02:
1.2 works but 1.3 doesn't

zuidwijk, 2020-08-02:
> 1.2 works but 1.3 doesn't

Yep here too! 1.2.0 works perfectly fine. 1.3.0 gives the error which I posted.

DavidDeSloovere, 2020-08-04:
FYI The Windows build works fine (after bypassing Smartscreen)

philletourneau, 2020-08-06:
I can't open it either, FWIW

dotWee, 2020-08-09:
Same here on macOS 10.15.6. Trying to open the application (`$ open ESPHome-Flasher-1.3.0-macOS/ESPHome-Flasher.app`) fails with this exception: 

> LSOpenURLsWithRole() failed with error -10810 for the file /Users/lukas/Downloads/ESPHome-Flasher-1.3.0-macOS/ESPHome-Flasher.app.

zuidwijk, 2020-08-09:
Found it! This solves it:

`chmod +x ESPHome-Flasher.app/Contents/MacOS/ESPHome-Flasher`

```
sh-3.2# ls -l /Applications/ESPHome-Flasher.app/Contents/MacOS/
total 30792
-rw-r--r--@ 1 marcel  staff  15765270 30 jul 11:29 ESPHome-Flasher
sh-3.2#
```

zuidwijk, 2020-08-09:
@dotWee Your error message gave me the clue :)

zuidwijk, 2020-08-09:
@OttoWinter can you fix this in the next/upcoming version?

dotWee, 2020-08-09:
Can confirm the application is just missing the executable flag. Thanks!

cmriquinha, 2020-10-27:
Hey, all I know it's been a bit since yall commented on this.  I have tried many versions.  Including ESPHome-Flasher (which will not open, per initial issue) as well as NodeMCU Pyflasher v.3 and v.4 apps not seeing usb serial port.  I am a bit of a loss and can not move on until I figure it out.  Any help would be appreciated as I am still unable to flash the esp8266 running on Mac Catelina.

suhlig, 2020-11-09:
```command
chmod +x /Applications/ESPHome-Flasher.app/Contents/MacOS/ESPHome-Flasher
```
fixes the problem.

zuidwijk, 2020-11-10:
@cmriquinha have you tried what @suhlig and I have mentioned as solution?

```
chmod +x /Applications/ESPHome-Flasher.app/Contents/MacOS/ESPHome-Flasher
```

jezmck, 2020-11-14:
I also had to allow the app to run in System Preferences > Security & Privacy > General (a button should appear at the bottom).

ronnyandre, 2020-12-04:
Still an issue.

ok-morpheus, 2021-01-03:
> ```shell
> chmod +x /Applications/ESPHome-Flasher.app/Contents/MacOS/ESPHome-Flasher
> ```
> 
> fixes the problem.

Great info.  But you guys are assuming the readers know to type that on a terminal command line.

IZ1IVA, 2021-01-05:
This fix unfortunately doesn't work on macOS 10.13.6 (tried on two different iMacs).

genebean, 2021-01-16:
Running the `chmod` command above via sudo in a terminal on BigSur and then allowing it to run from Security & Privacy worked for me

hmakmur, 2021-04-10:
I can confirmed that this does not work with MacOS High Sierra 10.13.6 despite adding exec bit as other has suggested. Those who got it to work, please specify what Mac OS version you are using. I wish there is a note about what minimum  MacOS is required to run this.

dotWee, 2021-04-10:
Just tested the latest release (1.3.0) on macOS Catalina 10.15.7, does not run without manually marking as executable.