esphome/esphome-flasher #29: Will not launch on mac os X

bakaufman, 2020-08-18:
Catalina 10.15.6
iMac (21.5-inch, Late 2013)
[image]
Fresh download today. 
No option to select in Settings>Security to bypass

rlust, 2020-08-18:
I have the same issue, will not open on iMac

davosian, 2020-08-18:
Same here on Catalina 10.15.6

davosian, 2020-08-18:
See https://github.com/esphome/esphome-flasher/issues/26#issuecomment-671061140 for a work-around

bakaufman, 2020-08-18:
So if you do the chmod (see other entry in github), you can get it to launch. Getting the Mac to see the device seems to be another matter.

davosian, 2020-08-18:
Did you go to System Preferences and allow this app in the Security and Privacy section? I think I had to do this if I remember correctly.

rlust, 2020-08-19:
I went to the Mac applications folder an ran 

chmod +x ESPHome-Flasher.app/Contents/MacOS/ESPHome-Flasher

This allowed it to run!

FrancoRivera, 2020-09-11:
I had the same issue, on Mac OS Catalina, @rlust fix worked perfectly

Just to reiterate:

1. Copy the ESPHome-Flasher.app to your Applications folder 
2. Open your Terminal (in usually in Applications -> Other -> Terminal)
3. Run chmod +x /Applications/ESPHome-Flasher.app/Contents/MacOS/ESPHome-Flasher
4. Go to your Applications folder and "Right Click" (two finger click or Ctrl+Click) and click on Open (Note: you have to right click the first time, **it  will not work if you simply double click**.)

It should run without a problem.

Stenvk, 2020-09-17:
Im still having trouble opening the flasher.
Im checking wether I can open the program before I buy all the stuff I need the program for excuse otherwise I wasted my money but I can't seem to get the program to open.
Also the terminal trick didn't work for me.

jrc, 2020-09-18:
Duplicate of https://github.com/esphome/esphome-flasher/issues/26

johnehowe, 2021-03-17:
Solution to do a Chmod from rlust and FrancoRivera works. See above. Thanks guys!