esphome/esphome-flasher #30: Getting virus warning from VirusTotal on ESPHome-Flasher-1.3.0-Windows-x64.exe

Tor-Adi, 2020-08-20:
[image]

Tor-Adi

sjude68, 2020-12-06:
Even I got a virus warning while downloading.

EugeneKorneev, 2020-12-23:
[2 engines detected this file](https://www.virustotal.com/gui/file/008f6aa24d287b91254d36f1187189a126817c74ae4ac5a99f0b27a2b38bd80f/community)
008f6aa24d287b91254d36f1187189a126817c74ae4ac5a99f0b27a2b38bd80f
ESPHome-Flasher-1.3.0-Windows-x64.exe

superkillared, 2021-03-19:
I got an even worse result testing the windows x86 version
[image]

sjude68, 2021-03-20:
Why is the virus file not yet removed.

OttoWinter, 2021-03-20:
This is a duplicate of other issues, I'm merging the discussions into #40.

See also https://github.com/esphome/esphome-flasher/issues/40#issuecomment-803265110 for more details. **These are false positives**.

closed 2021-03-20 by OttoWinter