esphome/esphome-flasher #34: serial port option states blank on Mac OS

evonashion, 2020-09-11:


kerflop, 2020-09-23:
Having the exact same issue.  I am on a Mac with USBC ports, so not sure if that is an issue.

MrVinceZ, 2020-10-03:
Do you plug the node directly to usb port ? 

Work fine with an usb hub between my MBP and the alimentation.

drice78, 2020-11-14:
Having this same problem. I am connected to directly to the USB port on my mac mini.

MrVinceZ, 2020-11-14:
> Having this same problem. I am connected to directly to the USB port on my mac mini. 

Maybe not enough power on USB port. Can you try with external alimentation ?

juicejuice, 2020-11-16:
Change USB cables too, some of my USB cables are power only and the data pins are not connected. It will only show the USB ports when a valid USB UART device is connected.

closed 2022-02-05 by github-actions[bot]