esphome/esphome-flasher #35: Got this error message on macOS 10.15.6

ranrinc, 2020-09-23:
macOS 10.15.6 with error result application cant be opened

[image]

help

SecondGrowth, 2020-09-23:
I had the same issue since yesterday.

SecondGrowth, 2020-09-23:
> I had the same issue since yesterday.

Just found from another thread that 1.2 works. Downloaded and opened successfully:

https://github.com/esphome/esphome-flasher/releases/tag/v1.2.0

ranrinc, 2020-09-24:
> > I had the same issue since yesterday.
> 
> Just found from another thread that 1.2 works. Downloaded and opened successfully:
> 
> https://github.com/esphome/esphome-flasher/releases/tag/v1.2.0

It work but I dont get USB info.. so did work either. :(

jesserockz, 2020-09-24:
See #29