esphome/esphome-flasher #36: no port

icekid67, 2020-11-03:
i get no serial port number for windows>

Jimothydoesntknow, 2020-11-13:
Downloading the drivers from this website helped me. 

https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers

bengomez31, 2020-12-06:
Downloading the Drivers Jimothydoesntknow stated worked for me

unknowngamer2005, 2020-12-17:
I installed the drivers but the port choosing window still is blank

unknowngamer2005, 2020-12-17:
> Downloading the drivers from this website helped me.
> 
> https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers

What sysytem are you using WIndows or Mac?

icekid67, 2020-12-17:
I have tried both, windows and Mac. Neither one will work.

Zixim, 2020-12-17:
The few times i got no serial port , is when I accidentally used a usb cable that doesn't support data transfer...

unknowngamer2005, 2020-12-19:
> I have tried both, windows and Mac. Neither one will work.
> […](#)
> Sent from my iPhone

Did you try updating CH340 driver? Because that was an issue i had.

OnDro7, 2021-05-07:
I had to install CH341SER driver ( http://www.wch.cn/download/CH341SER_ZIP.html ) and now it's working fine for me.