esphome/esphome-flasher #38: do not have permission to open application

popoybvargas, 2020-11-06:
You do not have permission to open the application "ESPHome-Flasher.app".
Contact your computer or your network administrator for assistance.

[image]

[currently on macOS Big Sur Version 11.0.1 Beta (2085022a)]

edwardscaleb9, 2020-11-08:
Same issue here on Big Sur 11.0 Beta

sogeniusio, 2020-11-10:
Getting the same thing.

aaronk6, 2020-11-14:
This is the same issue as discussed here: https://github.com/esphome/esphome-flasher/issues/26#issuecomment-671061140

So it’s actually not a Big Sur issue. Can be fixed by making the binary executable:

```
chmod +x ESPHome-Flasher.app/Contents/MacOS/ESPHome-Flasher
```

ctmfire, 2020-12-16:
when I try that it says, "no such file or directory"

aaronk6, 2020-12-17:
@ctmfire Then you’re in the wrong directory. Try `chmod +x /Applications/ESPHome-Flasher.app/Contents/MacOS/ESPHome-Flasher` (assuming you’ve moved it to the system-wide Applications folder).

ctmfire, 2020-12-26:
Thank you for the help.  It worked!

Syazvinski, 2021-04-23:
Ive converted the flasher to binary and it is opening, but its not finding my ESP under the COM ports section. Im using a new M1 mac, could this cause an issue.

wilddog6060, 2021-09-05:
> Ive converted the flasher to binary and it is opening, but its not finding my ESP under the COM ports section. Im using a new M1 mac, could this cause an issue.

I have the same problem - runs OK but cannot see ports.  Ran the Windows version in Parallels to see if it would help - no luck

poohzaza166, 2021-10-03:
i slove it by going in to the app and chmod the file under ./contents/macos/ change the permission to 775 and run it using ./ and it work

poohzaza166, 2021-10-10:
> permissions. When I tried typing the string of commands mentioned in the terminal I got the "No such file or directory" message. I have seen that people solve this but I don

no worries once you install the apps it will appear in Applications folder
[image]
once in you will saw the folder name contents under there you will saw the file structure
[image]
 here under MacOS you will see a file call ESPHome-Flasher
[image]
once in type

chmod 775 ESPHome-Flasher

austgresh, 2021-10-10:
@poohzaza166 that worked great to get the esphome flasher open, thank you for the clear instructions.  Now I have the same problem you mentioned right after with not being able to get a serial port com to show up.

austgresh, 2021-10-10:
ESP8266

poohzaza166, 2021-10-10:
can you be more specific like is it nodemcu or some random esp8266 from china?

what you could you is
1 check driver
2 check cable
3 test esp8266 if it dead or not

austgresh, 2021-10-10:
sorry nodemcu.  I'm trying to program led strips as a project.

poohzaza166, 2021-10-10:
i see just want to know are you using new m1 mac

austgresh, 2021-10-10:
Big Sur on an MB pro

poohzaza166, 2021-10-10:
ok is your nodemcu using ch210x chip or chg340x they way to check is on your board there will be a black chip near the usbport read it

austgresh, 2021-10-10:
210

poohzaza166, 2021-10-10:
ok plug it in and type

have you install ch210x driver?

austgresh, 2021-10-10:
I'm going to guess no.  Where should I get that?

poohzaza166, 2021-10-10:
https://www.silabs.com/interface/usb-bridges/classic/device.cp2102

download this CP210x Software package for Mac, includes VCP Drivers

austgresh, 2021-10-10:
installed the driver and still no com in flasher

poohzaza166, 2021-10-10:
did you reboot the system?

austgresh, 2021-10-10:
Just restarted and still no com

poohzaza166, 2021-10-10:
ok no worries