esphome/esphome-flasher #39: Have to specify the usb serial via CLI

jimmypuckett, 2020-11-27:
When I launch the app, it does not show the USB serial device in the GUI, so I have to pass it as a the `-p` flag in the CLI, which works...

```
$ ./ESPHome-Flasher -p /dev/cu.usbserial-AO009AL5 ~/Downloads/gate.bin
Using '/dev/cu.usbserial-AO009AL5' as serial port.
Connecting....
Detecting chip type... ESP32
Connecting...

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 24:0A:C4:C5:5E:A8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 9.5s
Compressed 15872 bytes to 10319...
Wrote 15872 bytes (10319 compressed) at 0x00001000 in 0.2 seconds (effective 530.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 143...
Wrote 3072 bytes (143 compressed) at 0x00008000 in 0.0 seconds (effective 1569.3 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4037.5 kbit/s)...
Hash of data verified.
Compressed 949360 bytes to 533857...
Wrote 949360 bytes (533857 compressed) at 0x00010000 in 12.8 seconds (effective 594.8 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!
```

jimmypuckett, 2020-11-27:
BTW - I am on an Apple M1 MBP (Silicon).  I tested with 1.3.0 & 1.20.

Normanras, 2021-03-26:
@jimmypuckett I tried your code in the CLI since my GUI also shows no serial port connection (but `ls /dev/cu*` confirms it is connected), but ./ESPHome-Flasher brings up no such file exists. I am in the correct directory and running `open ESPHome-Flasher` opens the GUI without issue. Any tips?

jimmypuckett, 2021-03-27:
@Normanras I was in the `/Applications/ESPHome-Flasher-1.2.0-macOS.app/Contents/MacOS` folder when tyrpign that.

You can do `/Applications/ESPHome-Flasher-1.2.0-macOS.app/Contents/MacOS/ESPHome-Flasher -p /dev/cu.usbserial-AO009AL5 ~/Downloads/gate.bin`

where `~/Downloads/gate.bin` is the path to your file & `/dev/cu.usbserial-AO009AL5` is the path to your adapter.

I hope that helps

Normanras, 2021-11-02:
@jimmypuckett Sorry I am only responding - I must have missed this. Yes, that is very helpful, thank you!

closed 2022-02-08 by github-actions[bot]