esphome/esphome-flasher #40: Trojan:Win32/Zpevdo.B  ???

qsiguy, 2020-12-14:
Windows Defender on a Windows 10 Home machine is reporting this trojan when I download the file below and removing it.  I have another copy of this file that I downloaded a few weeks ago that does not show as infected.  Has the file below on the repository been compromised?

ESPHome-Flasher-1.3.0-Windows-x86.exe

fabaff, 2020-12-14:
`clamav` which the latest signatures is not trigging an alert. 

```
$ clamscan ESPHome-Flasher-1.3.0-Windows-x86.exe 
/home/test/esphome-flasher/ESPHome-Flasher-1.3.0-Windows-x86.exe: OK

----------- SCAN SUMMARY -----------
Known viruses: 8951674
Engine version: 0.103.0
Scanned directories: 0
Scanned files: 1
Infected files: 0
Data scanned: 12.77 MB
Data read: 11.99 MB (ratio 1.06:1)
Time: 18.860 sec (0 m 18 s)
Start Date: 2020:12:14 12:19:25
End Date:   2020:12:14 12:19:44

$ clamscan ESPHome-Flasher-1.3.0-Windows-x64.exe 
/home/test/esphome-flasher/ESPHome-Flasher-1.3.0-Windows-x64.exe: OK

----------- SCAN SUMMARY -----------
Known viruses: 8951674
Engine version: 0.103.0
Scanned directories: 0
Scanned files: 1
Infected files: 0
Data scanned: 16.29 MB
Data read: 15.30 MB (ratio 1.06:1)
Time: 19.967 sec (0 m 19 s)
Start Date: 2020:12:14 12:31:41
End Date:   2020:12:14 12:32:01
```

It's a system tool after all and from the point of view from an AV solution most of those tools are harmful. The complete source code is [public](https://github.com/esphome/esphome-flasher/tree/master/esphomeflasher).

Zixim, 2020-12-17:
previous versions did not trigger the false positive... 
With 1.3.0 even the browser throws a temper tantrum when trying to save the download.

Have flagged the false positive through the Microsoft feedback mechanism...for all the good that will do.

EugeneKorneev, 2020-12-23:
[VirusTotal 17 engines detected this file](https://www.virustotal.com/gui/file/1db573038618bee45adefc3c05d1ab2a104fc5a728693ab3dbc6bb49205bff31/detection/f-1db573038618bee45adefc3c05d1ab2a104fc5a728693ab3dbc6bb49205bff31-1608575350)
1db573038618bee45adefc3c05d1ab2a104fc5a728693ab3dbc6bb49205bff31
ESPHome-Flasher-1.3.0-Windows-x86.exe

OttoWinter, 2021-03-20:
Hi,

Yeah I've see seen this before (https://github.com/esphome/esphome-flasher/issues/19, https://github.com/esphome/esphome-flasher/issues/9, https://github.com/esphome/esphome-flasher/issues/6, https://github.com/esphome/esphome-flasher/issues/3)

I can say with high certainty that this file is not a virus, the warning is a false-positive.

If there were a virus embedded, there could be several places in the build process where it's injected. I've analyzed all of them and haven't found any indication.

First the dependencies, wxPython and esptool are both widely used dependencies, and if there were an infection that these engines would now about, there would also likely be a CVE, which there is none.

Second the build server, when this first popped up I was building the app with my own computer. Could that have been infected? Unlikely because I almost never use the windows system, but still I converted to the app being built on GitHub (https://github.com/esphome/esphome-flasher/blob/master/.github/workflows/build.yml), but the warnings still showed up. So no dice there.

I've spoken to some colleagues at my university to figure out why the virus engines are going crazy. And I believe there's an alternative explanation. Note most the engine detections have "heuristics" in the name. esptool.py [bundles its own ESP bootloader in the python code here](https://github.com/espressif/esptool/blob/16703d0be03e1477741a856e2dcdc4896fca9267/esptool.py#L4171). Now bundling b64 binaries in python code likely is a common way for actual malware to obfuscate their source. So the engines' heuristics could be alerting to that even though the embedded bootloader is completely benign.

It's notoriously bad at figuring out why a malware engine alerts to something, so I see no way for figuring things out 100%. But if you want you can fork this project and run the GH Actions pipeline yourself to validate the released binary is free of any malware.