esphome/esphome-flasher #42: Unexpected error / AttributeError: 'RedirectText' object has no attribute 'isatty'

PieBru, 2020-12-19:
Hi, I'm on Arch Linux, where `esphome-flasher` installed flawless from AUR.
Trying to flash an ESP32 (MH-ET Live Mini) or a common ESP32 DevKit with ah ESPHome binary, the following error appears:
[image]

This is the console output:
```
$ esphomeflasher 
22:35:24: Warning: Mismatch between the program and library build versions detected.
The library used 3.0 (wchar_t,compiler with C++ ABI 1013,wx containers,compatible with 2.8),
and wxPython used 3.0 (wchar_t,compiler with C++ ABI 1014,wx containers,compatible with 2.8).
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/site-packages/esphomeflasher/gui.py", line 163, in run
    run_esphomeflasher(argv)
  File "/usr/lib/python3.9/site-packages/esphomeflasher/__main__.py", line 158, in run_esphomeflasher
    esptool.write_flash(stub_chip, mock_args)
  File "/usr/bin/esptool.py", line 2983, in write_flash
    print_overwrite('Writing at 0x%08x... (%d %%)' % (address + seq * esp.FLASH_WRITE_SIZE, 100 * (seq + 1) // blocks))
  File "/usr/bin/esptool.py", line 156, in print_overwrite
    if sys.stdout.isatty():
AttributeError: 'RedirectText' object has no attribute 'isatty'
```

Trying to proceed while waiting for a solution, i didn't find a proper and complete `esptool.py ` command to inizialize and flash a factory new ESP32 with an ESPHome binary.

Hints?
Thank you,
Piero

PieBru, 2020-12-19:
I bypassed the GUI problem using the CLI:
```
$ esphomeflasher --esp32 --bootloader bootloader_dio_80m.bin emon_xxxx.bin
```

I took the bootloader from:
```
/.platformio/packages/framework-arduinoespressif32/tools/sdk/bin/
```

ivlis, 2021-01-03:
I have exactly the same problem. It is a packaging problem with Archlinux.

BCNelson, 2021-01-24:
It looks like it can be tracked to arch's [python-wxpython](https://archlinux.org/packages/community/x86_64/python-wxpython/) being out of date

mozzwald, 2021-02-07:
I added this to main.py recommended in this [stackoverflow post](https://stackoverflow.com/a/47069974) and it now works for me:

``` import argparse
 from datetime import datetime
 import sys
+
+sys.stdout.isatty = lambda: False
+
 import time
 from esphomeflasher.common import open_downloadable_binary

henkiejan1, 2021-03-15:
I got exactly the same problem as TS. But then on Linux Mint Ulyssa.

More info:
Linux hj-Latitude-5420 5.11.6-051106-generic #202103111435 SMP Thu Mar 11 14:42:39 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

Python 3.8.5