esphome/esphome-flasher #47: USB serial driver for ESPHome Flasher for Big Sur

miguelarios, 2021-01-05:
I am having issues with flashing my [ESP32](https://www.amazon.com/gp/product/B0718T232Z/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1). The serial port that ESPHome Flasher is trying to use is cu.usbserial-0001 but it fails to connect.

I have tried to install https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers both the legacy and the normal one but I this didn't fix anything.

What driver is everyone using?

miguelarios, 2021-01-05:
here are my logs with the latest driver from here: https://www.silabs.com/community/interface/forum.topic.html/vcp_driver_for_macosbigsur110x-SWJK

```
Using '/dev/cu.SLAB_USBtoUART' as serial port.
Showing logs:
[15:04:00]1 01 0000f000 00001000
[15:04:00]I (75) boot:  2 factory          factory app      00 00 00010000 00100000
[15:04:00]I (82) boot: End of partition table
[15:04:00]I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1c404 (115716) map
[15:04:00]I (136) esp_image: segment 1: paddr=0x0002c42c vaddr=0x3ffb0000 size=0x02ed0 ( 11984) load
[15:04:00]I (141) esp_image: segment 2: paddr=0x0002f304 vaddr=0x40080000 size=0x00400 (  1024) load
[15:04:00]I (143) esp_image: segment 3: paddr=0x0002f70c vaddr=0x40080400 size=0x00904 (  2308) load
[15:04:00]I (153) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x71ff4 (466932) map
[15:04:00]I (324) esp_image: segment 5: paddr=0x000a2014 vaddr=0x40080d04 size=0x10b54 ( 68436) load
[15:04:00]I (363) boot: Loaded app from partition at offset 0x10000
[15:04:00]I (363) boot: Disabling RNG early entropy source...
[15:04:00]I (364) cpu_start: Pro cpu up.
[15:04:00]I (367) cpu_start: Application information:
[15:04:00]I (372) cpu_start: Project name:     esp32-cam-demo
[15:04:00]I (378) cpu_start: App version:      1
[15:04:00]I (382) cpu_start: Compile time:     Mar  5 2019 11:42:22
[15:04:00]I (388) cpu_start: ELF file SHA256:  88a3a0e577233ee2...
[15:04:00]I (394) cpu_start: ESP-IDF:          v3.3-beta1-506-gebdcbe8c6
[15:04:00]I (401) cpu_start: Starting app cpu, entry point is 0x40081100
[15:04:00]I (0) cpu_start: App cpu up.
[15:04:00]I (411) heap_init: Initializing. RAM available for dynamic allocation:
[15:04:00]I (418) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
[15:04:00]I (424) heap_init: At 3FFB8E68 len 00027198 (156 KiB): DRAM
[15:04:00]I (430) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
[15:04:00]I (436) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[15:04:00]I (443) heap_init: At 40091858 len 0000E7A8 (57 KiB): IRAM
[15:04:00]I (449) cpu_start: Pro cpu start user code
[15:04:00]I (20) cpu_start: Starting scheduler on PRO CPU.
[15:04:00]I (0) cpu_start: Starting scheduler on APP CPU.
[15:04:00]I (41) I2S: DMA Malloc info, datalen=blocksize=256, dma_buf_count=8
[15:04:00]I (41) I2S: PLL_D2: Req RATE: 39062, real rate: 39062.000, BITS: 16, CLKM: 16, BCK: 8, MCLK: 10000000.000, SCLK: 1249984.000000, diva: 64, divb: 0
[15:04:00]I (51) camera_xclk: PIN_CTRL before:3ff,39062
[15:04:00]I (51) camera_xclk: PIN_CTRL after:7fff
[15:04:00]E (4071) camera_demo: Camera probe failed with error 0x20001
```

jimmypuckett, 2021-01-08:
Is this a duplicate of #39?

mcaminiti, 2021-01-15:
Yes this appears to be the same issue as the other.  I just used the USB Serial device that showed up in the GUI with the command line command with -p flag and it worked just fine. 

cd /Applications/ESPHome-Flasher/Contents/MacOS
./ESPHome-Flasher -p /dev/cu.usbserial-0001 /Users/<username>/Downloads/aqua1_hub.bin

miguelarios, 2021-01-15:
Hey thanks. I think the difference with me was that it couldn't find the ESP despite the serial device being found under devices. What I found out was that to have the ESPHome GUI find my ESP was I had to hold the enable push button to wake up the ESP. This is how I was able to program. I think this is due to the module I bought.

alve89, 2021-01-27:
I found this solution: https://forum.arduino.cc/index.php?topic=702144.msg4793318#msg4793318

```
1.- Open ~/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/pyserial/serial/tools/list_ports_osx.py
2.- Comment out lines 29 and 30 and append these lines:
   iokit = ctypes.cdll.LoadLibrary('/System/Library/Frameworks/IOKit.framework/IOKit')
   cf = ctypes.cdll.LoadLibrary('/System/Library/Frameworks/CoreFoundation.framework/CoreFoundation')
```

That worked well for me!

closed 2021-11-06 by github-actions[bot]