esphome/esphome-flasher #51: ESPHome flasher error

MisterTFM, 2021-03-01:
Trying to use esphome-flasher to download a BIN file that was created using ESPHome under Home Assistant to a M5Stickc.

I get the following error(s) in the ESPHome-flasher log:
Using 'COM4' as serial port.
Connecting.......
Detecting chip type... ESP32
Connecting......

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-PICO-D4 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: YES
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 94:B9:7E:8D:92:28
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Unexpected error: Reading chip details failed: Timed out waiting for packet header

I can successfully download files using Arduino IDE using a baud rate of 115200.  I'm using Windows 10, x86, USB 2.0

Is there a way to select a slower baud rate when using ESPHome-Flasher?

SeByDocKy, 2021-04-28:
Got exactly the same error/problem with the builtin ESP32 of the maixiduno K210...

SeByDocKy, 2021-04-28:
Ok to fix it, I used the cmd lines option of esphomeflasher in linux : 

esphomeflasher -p /dev/ttyUSB1 --upload-baud-rate 115200 esphome_modelèfile.bin  

I tried first on windows, but cmd lines are no more working at least on 2 W10 I tried....  when typing : esphomeflasher -p COM6  --upload-baud-rate 115200 cat_detector.bin     nothing is done...

Drew765, 2021-05-06:
I've had the same issue as the OP a few times in the last few weeks on ESP8285's & 8266's. Never an issue before in 2+ years. Ended up using Tasmotizer 😒.
Would be happy to help in any way I can with further testing.

eranjo, 2021-05-28:
I had the same issue with my M5 Atom lite. Version 1.0.0 will work though:)

kiwipaulrob, 2021-06-23:
Same problem. Had to use ver 1.01 flash ESP32 for m5stick-C