esphome/esphome-flasher #52: Big Sur + macOS M1

jcastro, 2021-03-03:
Hello,
I have an iMac 2013 with Big Sur where I'm able to use esphome-flasher no problems. However connecting the same cable to my Macbook Air M1 I'm unable to get the USB serial port to show up. I've tried with tons of different cables and hubs, but no luck.

Is this a known bug?

Normanras, 2021-03-26:
I came to the issues section for the same problem! I can't get the serial port to show up in the app, however when I go to terminal and type `ls /dev/cu.*` it shows up without an issue. I can also confirm that I can transfer my code from the Arduino IDE and use the serial monitor.

HA-TB303, 2021-04-10:
I have the same issues. No serial interface available.

thenexus7, 2021-05-06:
Exact same thing... I had the app startup lock, resolved with the chmod method. On Arduino app you are able to see and flash, but ESPHome just don't see any ports

jcastro, 2021-05-31:
I'm now using NodeMCU Pyflasher and it works fine on M1 + Big sur

jrc, 2021-06-15:
No luck for me, but I discovered that the `esphome` CLI tool already offers the ability to flash OTA or via serial. It even has a "dashboard" subcommand which puts up a web UI.

srg74, 2021-08-10:
I search all possibilities to find fix for ESPhome flasher and no luck. Devs please pay attention it not working for M1 + Big sur for a long time.

Normanras, 2021-08-11:
If it is helpful to anyone on this thread, I keep this note next on my desktop to flash my ESP devices via the command line. I am also using an M1 Mac and have had very consistent, successful results with the command line. I gave up on using the GUI since it worked so often. Install ESPHome-Flasher as instructed. Ensure it is in your Applications folder.

Here is my note:

1.  `cd /applications/esphome-flasher.app/contents/macos`
2. `./ESPHome-Flasher -p /dev/cu/SLAB_USBtoUART ~/path-to-file/file-name.bin` 

- Make sure to change the ESP path and file path to the appropriate paths. For myself, the ESP is usually at `SLAB_USBtoUART` file path is `~/Downloads/example.bin`

Good luck! I hope this helps.