esphome/esphome-flasher #99: Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)

copenguin8, 2021-12-23:
I am just posting what it says and hoping you can help reply if more info is needed

Using 'COM3' as serial port.
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.........
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 78:E3:6D:0A:E7:D0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)

simonyel, 2022-02-22:
The same issue here, flashing espresence build 2.32

tko, 2022-04-03:
I read elsewhere that the `E9` is how a standalone(?) firmware should start with. However it looks like esphome generates a "combined" factory image or something, which starts with lots of `FF`s and the actual firmware starts at offset `0x1000`
I don't really know what to do with this information though. Seems the flasher is incompatible with the images produced by esphome for the moment.

mprowe, 2022-04-29:
... and me!

```
Using '/dev/ttyUSB0' as serial port.
Connecting...[07:42:18]ets Jun  8 2016 00:22:57
[07:42:18]
[07:42:18]rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
[07:42:18]waiting for download
.....[07:42:18][07:42:18]
[07:42:18]rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
[07:42:18]waiting for download
..Serial port closed!

Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting......
Detecting chip type... ESP32
Connecting.....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WD (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 98:F4:AB:75:7E:EC
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 16MB
Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)

```

michelcve, 2022-07-22:
Same here, anyone found a solution for this yet?

closed 2022-10-28 by github-actions[bot]