esphome/esphome-flasher #148: esphome-flasher 1.4.0 crashes on start in Ubuntu 22.04

goberhammer, 2022-06-26:
Downloaded latest 1.4.0, made executable.
When launched it crashes with this log:
```
$ ./ESPHome-Flasher-1.4.0-Ubuntu-x64.exec 
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgioremote-volume-monitor.so
/usr/lib/x86_64-linux-gnu/gio/modules/libgiognutls.so: undefined symbol: g_byte_array_steal
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgiognutls.so
/usr/lib/x86_64-linux-gnu/gio/modules/libgiognomeproxy.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgiognomeproxy.so
/usr/lib/x86_64-linux-gnu/gio/modules/libgiolibproxy.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgiolibproxy.so
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgvfsdbus.so

(ESPHome-Flasher-1.4.0-Ubuntu-x64.exec:10867): GLib-GIO-ERROR **: 17:21:11.181: Settings schema 'org.gnome.settings-daemon.plugins.xsettings' does not contain a key named 'antialiasing'
Trace/breakpoint trap (core dumped)
```
What can I do to help solving the issue ? Thanks :-)

sekt1953, 2022-07-03:
I do have the same problem, on a fresh installation of Ubuntu 22.04 LTS

sekt1953, 2022-07-03:
Hej
here some help If you disable Wayland and run X it will work
here is a guide to disable Wayland: https://github.com/sekt1953/Ubuntu_22.04LTS#how-to-enabledisable-wayland-on-ubuntu-2204-desktop

goberhammer, 2022-07-06:
Thanks seckt1953 for the suggestion. I didn't want to go back to X11 so for now I'm running the appimage in an old VirtualBox Ubuntu 20.04 VM with USB passthrough and it seems to work well.
I'll look into forcing the appimage to use XWayland, maybe it can solve the issue.

closed 2022-10-12 by github-actions[bot]