esphome/esphome-flasher #195: Cannot open Browse dialog on Ubuntu

aurelijusrozenas, 2022-12-14:
Hi,

esphome-flasher does not work on my machine, not sure what is the problem. The app opens initially and everything seems to be working unless I press "Browse".
```
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgioremote-volume-monitor.so
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgvfsdbus.so
/usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgioremote-volume-monitor.so
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Yaru/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Nutraukta (aborted) (core dumped)
```
Can this really because of some icon?

Running Ubuntu 22.10, 5.19.0-26-generic.
Downloaded binary from releases page, 1.3.0 and 1.4.0 acting the same.

niyazialpay, 2023-02-02:
I'm having same trouble

mmarsalko, 2023-04-23:
> 

fwiw, I worked around this by using the command line. eg:
```~/Downloads/ESPHome-Flasher-1.4.0-Ubuntu-x64.exec --port /dev/USB5 --esp32 ~/Downloads/plant2.bin ```